SofaPirate/Plaquette #4: Metronome?

sofian, 2018-01-27:
I'm thinking it would be a great add to the library to add a "metronome" (metro) object.

Example code (blink):

```c++
#include <Plaquette.h>

DigitalOut led(13);

Metro metro(1.0);    // sends one "bang" every second

void begin() {}

void step() {
  if (metro)
    led.toggle();
}
```

The object could offer the option to "repeat" a certain number of times.

Example: here we "blink" but only for 100 ms (0.1 seconds):

```c++
#include <Plaquette.h>

DigitalOut led(13);

Metro metro(1.0);    // sends one "bang" every second
Metro timer(0.1, 1); // times 0.1 seconds, repeats one time

void begin() {}

void step() {
  if (metro) {
    led.on(); // switch on
    timer.reset();
  }
  if (timer)
    led.off(); // switch off
}
```

Or:

```c++
#include <Plaquette.h>

DigitalOut led(13);

Metro metro(1.0);    // sends one "bang" every second
Metro timer(0.1); // times 0.1 seconds

void begin() {}

void step() {
  if (metro) {
    led.on(); // switch on
    timer.repeat(1+random(3)); // repeats 1, 2, or 3 times
  }
  if (timer)
    led.off(); // switch off
}
```

sofian, 2018-01-27:
I already wrote some code that I could share. What do you think @thomasfredericks ?

thomasfredericks, 2018-01-29:
@sofian Looks good. I prefer the ```timer.repeat()``` way. I prefer constructors with at most one argument: you can never guess what they do unless you look up the documentation.

sofian, 2018-01-29:
I agree wrt constructors however I don't see the bad in adding options for advanced coders.
What do you think of the name "Metro" for that object?

I was wondering, in terms of nomenclature, currently inputs are named with suffix "In" (AnalogIn, DigitalIn, StreamIn).

And the function generators are named with suffix "Osc". MetroOsc sounds weird.

We could choose the suffix "Gen" for every source that is a generator (SquareGen, MetroGen, etc).

Anyway these are just thoughts.

thomasfredericks, 2018-01-29:
Metro is a good name but is very different than a "signal generator" or an "Osc". The former sends events, and the later a constant signal. 

>  I don't see the bad in adding options for advanced coders.
Because the advanced coders make the examples, and then its back to square one.

sofian, 2018-01-30:
Ok.

A complementary, useful object would be a "Timer" object (a kind of Plaquette-compatible "Chrono") with a maxTime parameter that would output "0" until maxTime is reached and then "1" all along.

```c++
#include <Plaquette.h>

DigitalOut led(13);

Timer timer(1.0);

void begin() {}

void step() {
  if (timer) {
    led.toggle(); // switch on
    timer.reset();
  }
}
```

thomasfredericks, 2018-01-31:
The new timer is a kind of square osc with phase control :) Maybe all generators could use these methods.

sofian, 2018-01-31:
Yes that could make sense. If you think about it they could all have a period(), reset(), and repeat(n) methods.

sofian, 2018-01-31:
And also an invert() period.

sofian, 2018-06-15:
This has been implemented but still needs to be tested.

sofian, 2018-06-19:
I am closing it for now. I tested them.

closed 2018-06-19 by sofian