SofaPirate/Plaquette #5: Implement step() function as a single time unit

sofian, 2018-01-27:
I would like to discuss the principle ```step()``` function should correspond to a single time unit. This means that actions (put(), get(), etc) on objects during the ```step()``` function should have stable behavior within the step() scope. For example if I do:
```c++
void step() {
  float x1 = myInput.get();
  // ... some other operations ...
  float x2 = myInput.get();  
}
```

... then x1 and x2 should be equal. I think this would make good sense for inputs (sensors) and outputs (actuators).

Similary if I do:
```c++
void step() {
  myOutput.put(0.5);
  // ... some other operations ...
  myOutput.put(1.0);
}
```

... the put(0.5) will be ignored and only the put(1.0) will be operated.

Again: this goes with the idea that Plaquette's step() function works more like the Processing draw() function.

This, however causes a problem when we have filter objects such as normalizations:
```c++
void step() {
  myInput >> myFilter >> myOutput; // line 1
  1.0 >> myFilter; // line 2
}
```

In this case, what value will myFilter deliver on line 1? Options are:
* The value from last frame (makes sense in some cases but can be quite counter-intuitive in others)
* The immediate processed value (ie. how it works now, more intuitive but incompatible with what is suggested here)

@thomasfredericks ?

thomasfredericks, 2018-01-29:
I do not agree @sofian.  

> Again: this goes with the idea that Plaquette's step() function works more like the Processing draw() function.

Yes millis() will return the same value for the frame. But if the user changes a value, like the matrix for example, it is applied immediately and not at the end of draw.

If you do:
```
void step() {
  myOutput.put(0.5);
  // ... some other operations ...
  myOutput.put(1.0);
``` 
 the put(0.5) should not be ignored.

Return and use the immediate processed value. Otherwise, you also further break compatibility with other Arduino libraries.

sofian, 2018-05-29:
I believe for now we can preserve the status quo. Closing and labelling as "won't fix".

closed 2018-05-29 by sofian