SofaPirate/Plaquette #6: Add falling() and rising() to all DigitalIn objects

sofian, 2018-02-03:
Would be great to simply integrate that functionality into the DigitalIn object.
I wonder if the same could be said of AnalogIn... it is a bit more complicated in that case.

thomasfredericks, 2018-02-03:
Do you mean rose() and fell()? Outputs 1 on the step the value changed in the desired direction?

sofian, 2018-02-03:
Yes. In fact it would just be boolean functions that one could call to see if signal raised or fell.

sofian, 2018-02-03:
Eg.
```c++
DigitalIn button(2);

...
void step() {
  if (button.rose())
    led.toggle();
}
```

sofian, 2018-02-03:
Also: we could extend that feature to analog inputs and filter (ie. PqGetters). We could then add the possibility to call the function with a value (so that we will return true iff the value has changed by a certain amount).

Eg.

```c++
AnalogIn pot(A0);

...
void step() {
  if (pot.rose()) // this will be true if the value rose, even if by a tiny amount
  ...
  if (pot.fell(0.1)) // this will be true if the value fell by at least 0.1
  ...
}
```

sofian, 2018-06-06:
Also we should have an external object that would do the same thing but based on the values sent to it...

sofian, 2018-06-08:
I would suggest integrating these features as the high-level (ie. basically any unit that returns a value would be able to support this).

All objects would then have (at the very least) the following functions:
```c++
  virtual bool rose(float delta=0);
  virtual bool fell(float delta=0);
  virtual bool changed(float delta=0);
  virtual float previous() const;
  virtual float changeRate();
```

Here the "delta" specifies the rate of change (in value/second). This is to cope with the fact that values might have a bigger difference between two samples if the sample rate is lower.

Here are some consequences:
1. It requires adding an extra float in each object to preserve the previous value.
2. For digital objects (eg. DigitalIn) the "delta" would not be useful so it could be confusing.

Anyway this sketch is not final, but I would appreciate comments. @thomasfredericks

sofian, 2018-06-08:
Agreed.

sofian, 2018-06-08:
Then it would look like this in class PqDigitalGetter (which DigitalIn derives from):
```c++
  /// Returns true if the value rose since last call to step().
  virtual bool rose() { return changeState() > 0; }

  /// Returns true if the value fell since last call to step().
  virtual bool fell() { return changeState() < 0; }

  /// Returns true if the value changed since last call to step().
  virtual bool changed() { changeState() == 0; }

  /// Difference between current and previous value of the object since last call to step().
  virtual int8_t changeState() = 0;
```

One would then just need to override changeState() -- not sure of that name, but basically this is a function that returns (current - previous) values. So it should return -1, 0, or 1.

sofian, 2018-06-08:
Fixed, but needs testing.

commit fe786baa (2018-06-08): Added falling() and rising() to all DigitalIn objects (closes #6) (untested).
closed 2018-06-08 by sofian