SofaPirate/Plaquette #8: Add a ramping mechanism

sofian, 2018-02-04:
There needs to be a simple standard tool in Plaquette for "ramping" values.

Here is a suggestion:
```c++

Tween tween(1.0); //tweener that runs for 1 second
Tween tween2(2.0); // tweener that runs for 2 seconds

void begin() {
  tween2.easing(EASE_IN_SINE); // or perhaps TWEEN_IN_SINE
}

void step() {
  if (condition) {
    tween.start(); // this will "tween" between 0..1 for 1 second
    tween2.from(2).to(1).duration(2.5).start(); // this will "tween" from 2 to 1 in 2.5 seconds
  }
}
```

Easing functions reference: http://easings.net/

sofian, 2018-02-04:
Also it should be possible to interrupt a tween at any time:
eg.

```c++
tween2.to(1.5);
```

Not sure about the sequencing and method choices but this is the idea.

sofian, 2018-02-05:
I have created a first draft in a branch, find class ```Tween``` here:

https://github.com/SofaPirate/Plaquette/blob/feature/tween/src/PqExtra.h
https://github.com/SofaPirate/Plaquette/blob/feature/tween/src/PqExtra.cpp

thomasfredericks, 2018-02-05:
In Processing and Unity they use [lerp] instead of [tween]. 
* https://processing.org/reference/lerp_.html
* https://processing.org/reference/lerpColor_.html
* https://docs.unity3d.com/ScriptReference/Mathf.Lerp.html
* https://docs.unity3d.com/ScriptReference/Vector3.Lerp.html

In Pure Data and Max, they use  [line]  :
* https://docs.cycling74.com/max7/maxobject/line (go to a value in some time, from the current value)

Maybe something simpler like [line] would be better.

sofian, 2018-02-05:
Yes a lerp() method would be useful but since we are doing a lot of time-based interpolations it is useful to have an object that preserves the startTime to measure the duration. In fact this is what line() from my understanding.

sofian, 2018-02-05:
And once you have this then adding easing is kind of a freebie.

sofian, 2018-02-05:
Yes actually I have looked at line() and my Tween object does exactly that.

This is an example usage:

```c++
Tween myTween(1.0);

AnalogOut led(9);

void begin() {
  tween.fromTo(1.0, 0.0).start();
}

void step() {
  if (tween.isFinished())
    tween.to(randomUniform(0, 1)).duration(2.0).start(); // go to random value, always with duration of 2.0
  tween >> led;
}
```

I am not sure about the chained operations (.fromTo().duration().start()) as it might be confusing for some students. I have used it in other parts of Plaquette (basically whenever there is a function that changes a parameter). You don't *have* to use it, you can just call the methods separately, eg.:

```c++
myTween.fromTo(0.1, 1.);
myTween.duration(10);
myTween.start();
```

sofian, 2018-02-05:
Another option is to have different versions of ```start()```:

```c++
void start(float from, float to, float duration);
void start(float to, float duration);
void start(float duration); // potentially... this one would transition from 0 to 1
```

thomasfredericks, 2018-02-05:
I would just have:
```Line line(  <time to get there> , <optional initial value> ) ;```

Then you can call elsewhere:
```line.to( <new target value> )```

```line``` is event based. A signal equivalent would be a ```smooth```.

thomasfredericks, 2018-02-05:
What is your user case?

sofian, 2018-02-05:
User case: when pushing a button bring on an LED in 1 sec. When release switch it off in 0.2 secs.

```c++
if (button.rose())
  tween.to(1, 1);
else if (button.fell())
  tween.to(0, 0.2);

tween >> led;
}
```

I think it's important to be able to change the duration each time if wanted.

As for the "from" argument maybe we don't need it. However we should be able to set a new value without having to call ```to(value, 0)``` (ie. switch to value in 0 seconds).

thomasfredericks, 2018-02-05:
Why not stay in the same paradigm with:
```
button >> smooth >> led;
```

sofian, 2018-02-05:
No it is not the same thing. Smoothing refers to taking a signal and making it more smooth (ie. it is essentially a low-pass filter).

The idea of "tweening" or "lerping" is really to generate a time-varying "signal" that goes from one point to another point. It can be applied to many different contexts when you want a smooth transition.

thomasfredericks, 2018-02-06:
I am not convinced by your user case. 

There are already solutions in Plaquette to arrive to a similar result. 
You can not have real Tweening in Plaquette because Plaquette is for Physical Computing. There is 0% guarantee the output will be at the expected state. This is contrary to animation Tweening where things can jump from one location on screen to another.  

Maybe more feedback mechanisms would be more interesting.

sofian, 2018-02-06:
I still think we are missing a complementary solution for ramping between values. As of now, in terms of signal generators, all we have are oscillators ie. signals that repeat periodically. What you need in many circumstances is to vary smoothly the value of an output.

You could probably do the same thing using eg. a ```TriOsc``` but it is rather counter-intuitive.

Here is another use case. I have a servo motor (a big one!) that controls a door. To open the door I press a button. There is a movement detector on the other side of the door that detects that I have passed through the door and then abruptly shuts it.

```c++
DigitalIn button(...);
DigitalIn motion(...);
ServoOut servo(...);

Tween tweenServo(0); // initial value

void begin() {}

void step() {
  if (button.rose())
    tweenServo.to(1, 5.0); // 5 seconds to open the door
  else if (motion.rose()) {
    tweenServo.to(0, 1.0); // shut door in 1 seconds

  tweenServo >> servo;
}
```

Currently if we want to do this I do not know how we would do it -- at least not in a very simple way. And yes, of course, the servo might not *actually be* in the state we believe it will be in (because this is PhysComp) but I think this is besides the point. This is true of pretty much anything we do with Plaquette.

sofian, 2018-02-06:
I think this very simple interface would be a good start:

```c++
Tween(float startingValue);
void to(float endValue, float duration);
void fromTo(float startingValue, float endValue, float duration);
```
Not sure about the "fromTo" here ... but we need to be able to reset the starting point somehow. Another option instead of fromTo() would be to have a ```reset(float startingValue)``` function.

Other naming options:
```c++
Tween(float startingValue);
void tween(float endValue, float duration, <optional easingFunction>);
void tween(float startingValue, float endValue, float duration, <optional easingFunction>);
```
I'm not so sure about the word "tween" as it refers directly to animations -- tween stands for "inbetween" which is related to frames.

In the control/signal world we would use (as you suggested) the term "line" or "ramp" but this is confusing if we use non-linear easings (which I would like to integrate as they are quite "cheap" and open up possibilities in terms of expressivity).

If we were to use one of these this I believe I would prefer "ramp" to "line" as a ramp can be curved but a line is alway ... lin-ear!

Eg. 
```c++
Ramp(float startingValue);
void ramp(float endValue, float duration);
void ramp(float startingValue, float endValue, float duration);
```

Whatever name we have an alternative to using the same function name as the class is to use "start" or "begin" -- which represents that the action is just being started at function call.

```c++
void start(float endValue, float duration);
void start(float startingValue, float endValue, float duration);
```

thomasfredericks, 2018-02-06:
Seems good with start.
```
Ramp(float startingValue);
void Ramp::start(float to, float duration);
void Ramp::start(float from, float to, float duration);
```

sofian, 2018-03-11:
Done!

closed 2018-03-11 by sofian