SofaPirate/Plaquette #10: Rename random functions?

sofian, 2018-02-05:
I have decided to rename function ```map()``` to ```mapFloat()``` because otherwise it read confusing. In effect ```map(x, 1, 2, 3, 4)``` would return a float if x was a float and a long otherwise...

I also added some methods ```mapFrom01()``` and ```mapTo01()```.

I am tempted to correspondingly rename the ```randomUniform()``` functions to ```randomFloat()``` and ```random01()```. 

Comments?

sofian, 2018-02-06:
Je propose d'utiliser les suffixes "Float" pour exprimer qu'il s'agit d'un floating-point et "01" pour signifier l'intervalle [0, 1] qui revient souvent dans Plaquette. Ie.:

```c++
mapFrom01(x, toMin, toMax)
mapTo01(x, fromMin, fromMax)
mapFloat(x, fromMin, fromMax, toMin, toMax)
randomFloat(max)
randomFloat(min, max)
random01()
```

thomasfredericks, 2018-02-06:
> I have decided to rename function map() to mapFloat() because otherwise it read confusing. In effect map(x, 1, 2, 3, 4) would return a float if x was a float and a long otherwise...

Why does this bother you?

> Je propose d'utiliser les suffixes "Float" pour exprimer qu'il s'agit d'un floating-point et "01" pour signifier l'intervalle [0, 1] qui revient souvent dans Plaquette. Ie.:

The "01" is not very elegant. I would just do this:
```
unnormalize(x, toMin, toMax);
normalize(x, fromMin, fromMax);
map(x, fromMin, fromMax, toMin, toMax);
random(max);
random(min, max);
random();

```

sofian, 2018-02-06:
> Je propose d'utiliser les suffixes "Float" pour exprimer qu'il s'agit d'un floating-point et "01" pour signifier l'intervalle [0, 1] qui revient souvent dans Plaquette. Ie.:

>> The "01" is not very elegant. I would just do this:
>> ```
>> unnormalize(x, toMin, toMax);
>> normalize(x, fromMin, fromMax);
>> map(x, fromMin, fromMax, toMin, toMax);
>> random(max);
>> random(min, max);
>> random();
>> ```

I agree it is not particularly elegant and I would sure like a way to make it better.

I get the idea of the ```normalize()``` but the problem is that "normalize" is ambiguous ie. can mean different things. I also do not think the name is as obvious as mapTo01() and mapFrom01().

Perhaps we can use the term "unit" because in math the unit range refers to [0, 1]. So ```mapToUnit()``` ```mapFromUnit()```. However, I use "unit" to describe the components in Plaquette that can be sent to one another, so this makes it confusing once again. If we choose this solution then I suggest we find another name for Plaquette's units -- which is another discussion! Or we can use ```mapToUnitRange()``` ```mapFromUnitRange()``` -- but these function names are longer than simply calling ```mapFloat(x, 0, 1, min, max)``` :-)

thomasfredericks, 2018-02-06:
You could create a separate Mathf library. I think that would be a better solution.

```
Mathf.unnormalize(x, toMin, toMax);
Mathf.normalize(x, fromMin, fromMax);
Mathf.map(x, fromMin, fromMax, toMin, toMax);
Mathf.random(max);
Mathf.random(min, max);
Mathf.random();
```

`normalize()` means to conform to your system. So between 0 an 1.

`unnormalize()` to make unconfomm to your system. So not between 0 and 1.

Not sure about units. Normalization is more precise than unit.

sofian, 2018-02-06:
Let's drop the mapFrom01/mapTo01 discussion for now and focus on the mapFloat() vs map() to try and reach an agreement.

In the end I might be leaning towards overloading the names (ie. using map() and random() as you suggest). Getting an error from the compiler is still better than using the wrong function name.

I would like to do this for now. I am not sure about the Mathf library, I think operations like random() and map() are done quite frequently and should be part of the global scope. But I will think about it.

sofian, 2018-02-06:
So for now I suggest to fall-back to overloading functions ```map()```, ```random()```, ```random(max)```, ```random(min, max)```. People will have to deal sometimes with compiler errors.

Let's just think further about the question of the mapping from/to unit range. And the Mathf library.

sofian, 2018-02-06:
Oh! One alternative to using ```normalize()``` / ```unnormalize()``` would be to use ```standardize()``` / ```unstandardize()```. Conform to the standard ... or not -- as opposed to reference to the normal law. Some food for thought.

thomasfredericks, 2018-02-06:
> People will have to deal sometimes with compiler errors.

Oh no, please, that will fill up the issues with simple errors 👎 

Ok, so:

```
unnormalize(x, toMin, toMax);
normalize(x, fromMin, fromMax);
mapf(x, fromMin, fromMax, toMin, toMax);
randomf(max);
randomf(min, max);
randomf();

```

Normalize:
multiply (a series, function, or item of data) by a factor that makes the norm or some associated quantity such as an integral equal to a desired value (usually 1). 

The norm in Plaquette is between 0 and 1. Normalization has to do with mathematics. Standardization has to do with social and legal standards. It has nothing to do with mathematics. Also, normalize encompasses standardize (https://en.wiktionary.org/wiki/normalize) while being more precise.

thomasfredericks, 2018-02-06:
Another example:  https://processing.org/reference/PVector_normalize_.html

thomasfredericks, 2018-02-06:
```
unnormalize(x, toMin, toMax);
``` 
could be:
```
mapf(x, toMin, toMax);
```

sofian, 2018-02-06:
I don't like mapf() cause it uses an abbreviation. I prefer mapFloat() if we are going in that direction.

sofian, 2018-02-06:
But yes you are right about overloading map() we just cannot do that otherwise we will have compile errors when including Arduino libraries or running examples.

sofian, 2018-03-11:
Done. Still need to figure out mapTo/From01().

closed 2018-03-11 by sofian