SofaPirate/Plaquette #11: Yield equivalent

thomasfredericks, 2018-02-05:
Arduino for ESP8266 has a special function called ```yield()```. 

The ESP8266 must use the main thread to process WiFi paquets. By default, the code does this between calls to ```loop()```. But if the user requires some code that locks the main process, the user can make calls to ```yield()``` so that the ESP8266 does the behind the scene stuff.

Something similar for Plaquette would be very helpful for compatibility with third party libraries.

Note that Arduino for ESP8266's ```delay()``` integrates the ```yield()``` function so that ```delay()``` does not lock-up the WiFi.

sofian, 2018-02-05:
Do you suggest for example that we call ```yield()``` in Plaquette.update() on ESP8266-ESP32 such that if there is a lot of things to process then we do not block the process?

thomasfredericks, 2018-02-05:
```yield()``` is automatically called by ```loop()``` in Arduino for ESP8266. Therefore Plaquette already calls it.

I do not think the ESP32 needs a ```yield()``` because it has a separate processor to manage the WiFi and Bluetooth network.

I am suggesting a ```Plaquette.yield()``` so that Plaquette's internal private ```step()``` can be called from elsewhere.

sofian, 2018-02-05:
So the yield() would essentially call step()?

thomasfredericks, 2018-02-05:
Not step() but the internal step() i.e. ```Plaquette::update();```

thomasfredericks, 2018-02-05:
In a way, to make Plaquette::update() public.

sofian, 2018-02-05:
So one would call it using ```Plaquette::update()``` ? Wouldn't it be nicer to have some kind of a global function like on the ESP8266 (however ```yield()``` is already taken so we should use something else)?

thomasfredericks, 2018-02-06:
> So one would call it using Plaquette::update() ?

yes

>  Wouldn't it be nicer to have some kind of a global function like on the ESP8266 (however yield() is already taken so we should use something else)?

yes, but not sure I would make it global as it is an advanced feature.

sofian, 2018-02-06:
In fact ```Plaquette::update()``` _is_ public so you can already use it like you are intending to.

thomasfredericks, 2018-02-06:
Ok. I did notice that. As long as it stays in the public API.

closed 2018-02-06 by thomasfredericks