SofaPirate/Plaquette #9: Develop a mechanism for reentrant filters

sofian, 2018-02-05:
It would be nice to be able to use the piping system to process stateless/reentrant functions of the kind ```y = f(x)```. For example:
* f(x) = -x
* f(x) = 1/x
* f(x) = a x + b
* f(x) = map(x, l1, h1, l2, h2)

Ideally this should be easily and quickly done using predefined objects. Eg. (this is just an example):

```c++
in >> Pq.invert() >> out
```

sofian, 2018-02-05:
Currently it is only possible to apply operations on the first operand:

```c++
- (2 * in + 3) >> out;
```

So if there is a filter in the middle it is possible to apply operations by putting the expression in parenthesis but it is kind of weird:

```c++
- (2 * (in >> normalizer) + 3) >> out;
```

It would be probably more natural to do:

```c++
in >> normalizer >> func("-2*x + 3") >> out;
```

... but of course we do not want to have to support this kind of "inside script". Still perhaps something like:

```c++
in >> normalizer >> Pq.mult(2) >> Pq.add(3) >> Pq.mult(-1) >> out;
```

Similar to what you would find in a dataflow system eg. Pd:

```
[ in ] --> [ normalizer ] --> [ * 2 ] --> [ + 3 ] --> [ * -1 ] --> [ out ]
```

I am really not sure about that whole idea btw but this is something I have been thinking about alot.

thomasfredericks, 2018-02-05:
That is next level...

sofian, 2023-03-16:
Closing this issue: it could go in another library (perhaps) but should not be core.

closed 2023-03-16 by sofian