SofaPirate/Plaquette #12: Add tone() & noTone() functions to DigitalOut

sofian, 2018-02-06:
Arduino implements tone() and noTone() functions on digital outputs which use timers to generate square sound waves. These are functions are useful especially for beginners wanting to play with basic sound. Plaquette's oscillators are too slow for sound generation purposes -- although this might be different with faster chips.

https://www.arduino.cc/reference/en/language/functions/advanced-io/tone/
https://www.arduino.cc/reference/en/language/functions/advanced-io/notone/

thomasfredericks, 2018-02-06:
Tone is so basic and steals a timer from the servos. I would not implement it because there are better solutions to sound playback and sound synthesis on amateur chips is at our doorstep.

sofian, 2018-02-06:
I agree that it is a cheap trick for amateurs. I do not like these functions and I do not use them personally. But I recognize that it provides an easy way for beginners to start working with sound.

* Buzzers are very simple to hook up and it provides an alternative output to LEDs.
* There are many basic-level projects that use tone() such as alarms, pianos, and melodies.
* Whether we like it or not it *is* part of the Arduino library and unless we provide an easy alternative, beginners will want to use it. It is a missing feature in Plaquette, something that Arduino has and we don't.
* Any alternative -- at least to my knowledge -- will use one of the PWM timers.

In any case, I recognize that we should offer, if possible, a better solution to sound playback -- in fact, for fast-paced signal generation -- as part of Plaquette's built-in functionalities. It should be a user-friendly solution -- something *not* like Mozzi for example. But what is that solution if it is not tone() / noTone()?

thomasfredericks, 2018-02-06:
Tone already breaks Arduino (by breaking Servo). 

If the user wants, he can break Plaquette also simply by using the Tone command. No need to pollute Plaquette with it.

The examples with Tone are crap because there is no amplification (like the examples for Servos that do not provide proper power supplies). Students get 0 gratification because they cannot hear the sound properly. You can get the same effect with a digital buzzer (not piezo). 

Crap is crap and is never missed 💩

thomasfredericks, 2018-02-06:
Basically, there is no proper solution for sound with Arduino.

sofian, 2018-02-07:
Ok so the decision is to leave that crap out of Plaquette. Closing this issue. 💩

closed 2018-02-07 by sofian