SofaPirate/Plaquette #15: Manage problems related to undefined sampling rates for smoothings and other filterings

sofian, 2018-03-11:
In the current implementation, smoothings are performed using moving averages that take samples with an undefined sample rate (basically as fast as the step() function goes). This is problematic: basically we have signals read at undefined sample rates.

The main issue here is that the same smoothing factor will have a different effect if the step() function takes more or less time to process, which can make it difficult to create stable and modular code in Plaquette. I believe there are ways to improve the behaviors of the filters without hampering the user-friendlyness of the interface.

We need to have a discussion concerning sampling rates in Plaquette and how these are to be handled.

thomasfredericks, 2018-03-11:
As microcontrollers are evolving , better solutions are appearing. The ESP8266 already allows timers (that do not break other functions). I would look at what are the developments for the ESP32 as we can agree that the only good boards are the Teensy or the ESP*

thomasfredericks, 2018-03-11:
@sofian check this fixed timing library for ESP8266 : https://github.com/esp8266/Arduino/tree/master/libraries/Ticker

sofian, 2018-03-12:
Timers is one way to implement the sampling rate at a regular rate. And I do agree that Plaquette should truly be targeted at Teensy and ESP controllers which have much better CPU frequency/speed and can process floating point operations efficiently.

But what I have in mind here goes beyond this: it is truly a question of API.

I want to be able to define smoothing/normalizing factors as a value in seconds which corresponds or approximates the "window" over which the signal is smoothed (in an exponential moving average this is not a "true" window but really an approximation):
```c++
Smoother smoothOverLastSecond(1.0); // smooth signal over a sliding window of 1 second
Smoother smoothOverLastMinute(60.0); // smooth signal over a sliding window of 1 minute
```

thomasfredericks, 2018-03-12:
Yes. But you need a timing system to define the window. The ESP8266 has a method of providing this fixed time window.

sofian, 2018-03-12:
Not necessarily. If your sampling rate is lower than the speed of your step() you can just ignore samples during update() until the time unit is reached. Of course this is imperfect but probably good enough for low sampling rates (which in most cases of interaction design with humans are quite okay, cause human perception happens mostly at relatively low frequency).

sofian, 2018-03-12:
This being said I agree that we should aim the Teensy and the ESP* families as our target platforms. But it remains that Arduino compatibility is still important -- and possible in the short run. We could support all these platforms, but on the Teensy / ESP* make processing better / crisper.

sofian, 2018-05-07:
What I suggest is that we introduce a "sampling rate" concept in Plaquette. Here is a suggested interface:

```c++
/// Sets the sampling rate as n. samples per seconds and returns it.
float samplingRate(float hz);

/// Returns the current sampling rate.
float samplingRate();
```

The only issue is that if the processing time does not allow for the chosen rate (cause too much computation is happening) then things will implode. A solution is to calculate the *actual* sampling rate and have ways to access it, for example:

```c++
/// Returns the actual sampling rate of the last iteration. Will always be >= samplingRate().
float realSamplingRate();
```

I think that with these tools we can regularize the normalization processes so that they become more-or-less invariant to CPU speed / computation time / etc.

thomasfredericks, 2018-05-07:
The Mozzi library has two sampling rates. The CONTROL_RATE and the AUDIO_RATE. I believe it recycles the timer used for Arduino's  millis() micros() delay() and delayMicroseconds() .

https://sensorium.github.io/Mozzi/

Since the timer will be in interrupt mode, things should not implode.

sofian, 2018-05-07:
They can still implode if you consume too many resources. The use of timers just prevents shit from happening because of delays() and other blocking things like that.

In any case: I think we should aim for "timer support" on platforms such as the ESPs and Teensy but this needs a more extensive discussion to make it work properly.

sofian, 2018-05-29:
I have made some progress here. The new version of Plaquette runs at a definite sampling rate that the user can change by calling sampleRate(float). For now we are just "waiting" between each step. However by allowing a "manual mode" (which would be super easy by adding an option to include a <PlaquetteManual.h> or <PlaquetteLib.h>) you will be able to use your own system if you want -- e.g. use timers.

sofian, 2018-05-29:
Should we say "sampleRate" or "samplingRate" btw...?

thomasfredericks, 2018-05-29:
> Should we say "sampleRate" or "samplingRate" btw...?

sampleRate

> However by allowing a "manual mode" (which would be super easy by adding an option to include a <PlaquetteManual.h> or <PlaquetteLib.h>) you will be able to use your own system if you want -- e.g. use timers.

And what does `sampleRate(float)` do in manual mode? Maybe there should be a parent for Plaquette that manages this sampleRate and if we don't want it, we use the child:
* `PlaquetteAuto` -> parent that replaces `loop` and `setup`
* `Plaquette` -> child with no control on the sample rate

sofian, 2018-05-29:
Yes I think sampleRate() would only be available in the "auto" mode.

thomasfredericks, 2018-05-29:
Are you taking into account how slow analog reads are? This is a real problem with fast sample rates. Especially with all the floating point math. You should play around with Mozzi and see how this is a real issue.

Instead of using a fixed sample rate, you could use the method in Unity: use a counter for the number of elapsed micros and us that as a multiplier for the filters and oscillators.

thomasfredericks, 2018-05-29:
> Instead of using a fixed sample rate, you could use the method in Unity: use a counter for the number of elapsed micros and us that as a multiplier for the filters and oscillators.

This would also solve the discontinuity between the `auto` and `lib` mode.

sofian, 2018-05-31:
Done.

commit 4407fb51 (2018-05-31): Allow specification of smoothing / averaging factor in terms of seconds rather than "alpha" (closes #15).
closed 2018-05-31 by sofian