SofaPirate/Plaquette #17: What to do with the = operator?

sofian, 2018-03-11:
Currently, in Plaquette, the proper way to assign values to outputs is through the piping ``>>`` operator. Yet, because the inputs/outputs can be used "as" variables such as:
```c++
DigitalIn in(2);
DigitalOut out(13);
...
if (in)
  in >> out;
...
```

Some users are confused and think they can do things like:
```c++
out = 1;
in = 0;
```

Incidentally, this currently *works* but does not do anything.

1. Make operator= private for inputs

We need to at least make the operator = private for inputs so that
```c++
in = 0;
```

will produce a compilation error.

2. What about outputs?

However, we need to decide if 
```c++
out = 1;
```

is allowed or if we should make this also private.

In mbed (https://www.mbed.com/en/) this is allowed. The problem for me is that the "=" sign is usually an assignation operator. The problem is that the put() function can decide to filter out the value, leading to strange behaviors such as:
```c++
DigitalOut led(13);
...
led = 200; // the put() function will set the output to "1" not to 200
if (led == 200) // this will never trigger
...
```

or things that just don't really make sense:
```c++
Smoother smoother;
...
smoother = 3; // does not make sense
```

My recommendation is thus to make all of the = operator private so as to prevent users from using it. What do you think @thomasfredericks ?

thomasfredericks, 2018-03-11:
@sofian yes. = is too confusing.

sofian, 2018-03-12:
Ok. I'll just forbit its usage then.

commit a01bee10 (2018-03-12): Forbid assignations using operator= (closes #17).
closed 2018-03-12 by sofian