SofaPirate/Plaquette #18: errors compiling for NodeMCU 1.0

davidm-public, 2018-05-02:
excellent library!
does it support ESP8266 ?

sofian, 2018-05-02:
It should. Could you give more details about the compilation errors and the platform you are using?

sofian, 2018-05-02:
Thanks for your feedback!

davidm-public, 2018-05-04:
nope, hasn't work for me on esp8266, or Digispark, btw.
Does work on OEM Uno just fine.
Will try again and report errors, really like your abstractions.

sofian, 2018-05-05:
This should fix the problem: https://github.com/SofaPirate/Plaquette/commit/0369a6d8d79882bcfac51acad99ce79eb963cb34

I have pushed the changes and created a new version (0.1.7): you should be able to upgrade from the Library Manager -- probably need to wait 1 hour or so until Arduino catches up the changes.

sofian, 2018-05-05:
Thanks. That would help a lot!

davidm-public, 2018-05-07:
I can confirm that this code works on the Ardunino Uno (clone), and ESP8266 (NodeMCU 1.0 ESP13-E).

```cpp
 #include <Plaquette.h>

// Analog output.
AnalogOut led(5);

// Main sine oscillator (period of 1 sec).
SineOsc oscillator(1.0);

// Sine oscillator with period of 10 sec used to modulate oscillator.
SineOsc modulator(10.0);

void begin() {}

void step() {
    oscillator.period(0.5 + 2*modulator) >> led;
}
```

sofian, 2018-05-28:
Hurray!

closed 2018-05-28 by sofian