SofaPirate/Plaquette #23: Add support for OSC (through Masage)

sofian, 2018-05-29:
This is very important and probably needs a good discussion on how it should be done.

The way I see it, here is an example of code that would apply this in Plaquette:
```c++
AsciiMassageIn sensor("/signal");
AsciiMassageOut output("/output");
Smoother smoother;

...

sensor >> output;
```

In other words each object would be able to receive (or send) one floating point value to a specific address string.

Maybe this is too limited?

@thomasfredericks I am not familiar with the latest way things are done currently in Massage. Is it possible to use it to send and receive OSC messages over UDP for example?

thomasfredericks, 2018-05-30:
In recent tests with wireless micro-controllers, I also advise against UDP as UDP does not guarantee data reception and this is a major issue in physical computing.

sofian, 2018-06-02:
Ok I will start with Massage integration then.

I think we will create a separate library for this (PlaquetteAsciiMassage, PlaquetteSlipMassage).

closed 2025-05-15 by sofian