SofaPirate/Plaquette #32: Not sure smooth is debouncing

thomasfredericks, 2018-06-08:
In the **Pq_02_Button.ino** example, there is the following line:
`button.smooth(); // debounce button`

Smoothing is not the same as debouncing. For example, if the the input oscillates quickly between `0` and `1`, then the smoothed value will also oscillate between `0.51` and `0.49`. If the threshold is `0.5`, this will result in false triggers. 

Debouncing works in relation to the *stability* of the state over time. It needs a time argument and only outputs the final state if the input is has been stable for that time.

sofian, 2018-06-08:
I agree. Is this something you could implement?

thomasfredericks, 2018-06-08:
> I agree. Is this something you could implement?
Yes. But how does this work with `smooth`? Should `smooth` be there?

sofian, 2018-06-08:
Good question. I would say: probably not.
People who really want to smooth anyway can use the Smoother class.

sofian, 2018-06-08:
But for now just add the debouncing: I can take care of removing the smoothing.

sofian, 2019-11-05:
Done.

closed 2019-11-05 by sofian