SofaPirate/Plaquette #33: Should we be able to use "analog" units as booleans?

sofian, 2018-06-11:
Plaquette allows to use digital inputs as booleans eg.

Example 1:
```c++
DigitalIn button(2);
...
void step() {
  if (button)
    // do something
}
```

It also allows to convert booleans into a value eg.
Example 2:
```c++
DigitalIn button(2);
AnalogOut pwmLed(9);
...
void step() {
  button >> pwmLed; // in this case pwmLed will be set to 0% or 100% depending on button state
}
```

Also it supports directly converting float values to digital:
Example 3:
```c++
AnalogIn input(A0);
DigialOut led(13);
...
void step() {
  input >> led; // in this case led will be set to 0% if "iinput < 0.5" and to 100% otherwise
}
```

Conversely it allows this:
Example 4:
```c++
AnalogIn input(A0);
...
void step() {
  if (input) // will be true if input < 0.5
    led.on();
  else
    led.off();
}
```

QUESTIONS:

A. Should Plaquette allow to treat float-based units (analog) as booleans (example 4)
B. Should Plaquette allow the automatic conversion of digital into analog (example 3)

My first take on this is that the automatic conversion to boolean in Example 4 is a bit weird and might be counter-intuitive so I would tend to answer (A) as "No". However if we do this then it seems like we should also get rid of Example 3 and answer (B) as "No". What do you think @thomasfredericks ?

thomasfredericks, 2018-06-11:
No to both.

There is already a standard that applies to the conversion of numbers to Boolean (digital) values:
```
0: false
anything else: true
```

Example 3 should therefore be:
```
AnalogIn input(A0);
DigialOut led(13);
...
void step() {
  input >> led; // in this case led will be set to 0% if "input == 0" and to 100% otherwise
}
```

Example 4 should therefore be:
```
AnalogIn input(A0);
...
void step() {
  if (input) // will be true if input > 0
    led.on();
  else
    led.off();
}
```

In 5V logic signal voltage, a digital input does not switch between LOW and HIGH at 2.5V. But rather uses a high and low threshold with a dead zone: https://www.allaboutcircuits.com/textbook/digital/chpt-3/logic-signal-voltage-levels/

The proper way to do example 4 would be:
```
Theshold thresh( 0.25 , 0.75 ); // Threshold( low limit , high limit)
AnalogIn input(A0);
...
void step() {
  if (input >> thresh ) // will be true if input > 0.75 and false if input < 0.25. Otherwise state stays the same
    led.on();
  else
    led.off();
}
```

sofian, 2018-06-11:
Ok. I agree. Your example with the logic signal voltage clearly reinforces the fact that we should not presume any kind of automatic conversion from float to bool and vice versa. The user should make it explicit.

Except when you say:

> Example 4 should therefore be:
AnalogIn input(A0);
...
void step() {
  if (input) // will be true if input > 0
    led.on();
  else
    led.off();
}

IMHO we should just not allow it - ie. it should yield a compile error. This "proper" way to do things is actually kind of a shitty side-effect of the permissive nature of C++ which conflates logic with numeric. Let's just get rid of it and force people to specify clearly what they want to do when it comes to floats.

closed 2019-11-08 by sofian