SofaPirate/Plaquette #85: DigitalOut example - SINK and SOURCE

eringee, 2023-12-28:
For the online documentation http://sofapirate.github.io/Plaquette/DigitalOut.html

The introduction of the SINK and SOURCE mode to beginners can be confusing and also unnecessary, since DigitalOut defaults to SINK mode anyways.

Recommendation: in the example code and the documentation, instead of this:
```
#include <Plaquette.h>

DigitalOut led(13, SINK);
```

Do this:

```
#include <Plaquette.h>

DigitalOut led(13); 

//it is possible to operate the DigitalOut in SOURCE mode, for example if you are powering a common anode RGB LED
//DigitalOut led(13, SOURCE); 
```

Another suggestion is to replace the potentially confusing terms of SINK and SOURCE with terms like DIRECT and INVERTED.

sofian, 2025-02-12:
I see what you mean. It is true that DIRECT and INVERTED might be more clear. It would follow from #62 to stick to those terms. @thomasfredericks any opinion on this?

thomasfredericks, 2025-02-12:
@sofian, sure, as long as the distinction with the INTERNAL_PULLUP is clear. Maybe as another argument?

sofian, 2025-02-13:
@thomasfredericks SINK and SOURCE only concern DigitalOut, and are basically numerically equivalent to DIRECT and INVERTED. See PqCore.h:
```
enum {
  DIRECT,
  INVERTED,
  INTERNAL_PULLUP,
  SOURCE = DIRECT,
  SINK = INVERTED
};
```

So all we could do is deprecate SINK and SOURCE and update the documentation.

thomasfredericks, 2025-02-13:
My comment takes into account #62 
Seems I wrote the same comment in Jan 9 2023.
Inversion can and maybe should be seperate from the internal pullup state.

sofian, 2025-02-14:
I see, yes, I guess we never settled on this.

I would keep DIRECT, INVERTED and INTERNAL_PULLUP for the following reason: 

The concept of the direct/indirect mode refers to the concordance between the physical signal and the "on"/"off" status of the software unit. In other words, in the case of digital inputs, it refers to whether the value of the input signal is high (5V) (DIRECT) or low (GND) (INVERTED) when the unit is considered to be "on"; when we use an internal pullup, then automatically we are in a situation where "on" corresponds to GND.

In other words, adding INTERNAL_PULLUP as a separate argument would allow someone to declare a unit in DIRECT mode with an internal pullup, which does not make sense -- at least according to that definition.

sofian, 2025-02-14:
But I see your point, I understand the possible confusion, now that all of our in/out units all share the DIRECT/INDIRECT keywords, the INTERNAL_PULLUP seems like a special case just for inputs.

But making the internal pullup a separate "flag" will lead to a very confusing situation IMHO, one that will happen very frequently:

```c++
DigitalIn input(2, DIRECT); // direct is the default
DigitalOut led(13, DIRECT);

void begin() {
  input.enableInternalPullup();
}

void draw() {
  input >> led; // <-- ceci va faire que la LED va être éteinte quand on appuie sur le bouton car pour un circuit avec le input pullup, la "pin" est à 5V par défaut et tombe à GND lorsqu'on pèse sur le bouton
}
```

So I would keep it as is for this reason.

commit 3fa81359 (2025-03-03): Deprecated SOURCE and SINK in favor of DIRECT and INVERTED (closes #85).
closed 2025-03-03 by sofian