SofaPirate/Plaquette #136: Ambiguous Width for Waves

thomasfredericks, 2025-06-06:
For the `Wave` classes, the `width` parameter is ambiguous. It is probably named in reference to **pulse width modulation**, but that term is specific to **square/pulse waves**.  

In older oscillators that could generate multiple waveforms, the  **pulse width modulation** would **only** modulate the **square/pulse wave** and would have no effect on the other waves.

Modern oscillators use a more generic term like **skew** to indicate if the wave is _skewed_ towards the left or the right. 

A square/pulse wave skewed to the left (the SKEW H knob controls the horizontal skew) :
[image]

A sine wave skewed to the left (the SKEW H knob controls the horizontal skew) :
[image]

www-mariie, 2025-06-07:
I agree that the term "width" is ambiguous and i think it would be a good idea to change it. I like "skew". Alternatively, Touch Designer uses the term "bias" with the square wave pattern, but i think "skew" is better.

eringee, 2025-06-13:
I like skew much better, it repeats language from other softwares like in photo editing when you skew the image, so if a beginner saw the word they might intuitively know what it means.

sofian, 2025-06-14:
I like skew too. Seems like we all agree. I'll change it. @www-mariie will have to update the images for the documentation as well.