SofaPirate/Plaquette #138: Metronome and AbstractWave repetitive code

sofian, 2025-06-13:
There is a lot of code repetition between Metronome and AbstractWave as they share many common points in terms of timing, repetition, etc. although they are different in nature. There are different ways to approach this but I think it would be worth looking into it to save code and headaches.

thomasfredericks, 2025-06-14:
Yes @sofian. I think it is important that both use `phaseTimeUpdate` (found in [Plaquette/src/pq_osc_utils.cpp](https://github.com/SofaPirate/Plaquette/blob/master/src/pq_osc_utils.cpp)) so that all objects are tightly synchronized.

Maybe all objects should use `phaseTimeUpdate` for timing instead of alternative methods—especially those that do not account for "lost" milliseconds, like:

```cpp
if (millis() - start >= interval) {
    start = millis();
    ...
}

thomasfredericks, 2025-06-14:
Maybe a `Phasor` helper object could keep track of `phaseTime`, `phase` and overflow? (see #139 about naming)