SofaPirate/Plaquette #139: Mass internal find and replace renaming of `phaseTime` and `fixed_t` to more descriptive names

thomasfredericks, 2025-06-14:
I find the name `fixed_t` is too confusing. It is actually a  **Q0.32**. The [Q number format](https://en.wikipedia.org/wiki/Q_(number_format)) indicate 0 bits for the integer part and 32 bits for the fractional part. Q formats are often signed, even  **Q0.32**. So the full value range of a Q0.32 signed integer is approximately [-1.0, 1.0), not [0, 1). So its unsignedness should be indicated.

I propose renaming it to something more explicit, such as:
- `Q032u_t`
- `Q0_32u_t`

I find the name `phaseTime` is also confusing, as it mixes many concepts (phase, time and a Q format). I think it would help if all values in  **Q0.32** format were clearly tagged as such in their name:
- `phaseTime32` 
- `phasor32` (the term *phasor* is used in Pure Data to mark time as a percentage)

Note that I would recommend all properties in **Q0.32** be tagged with the suffix `32`.

sofian, 2025-06-16:
Sure, we can do it. I suggest using `q0_32u_t` for the type and `phaseTime32` for the phase time.

sofian, 2025-06-16:
Actually we should rename `phaseTime32` simply `phase32` and rename appropriate functions accordingly (it's not about time, just about phase).