SofaPirate/Plaquette #141: fixedTofloat rounding error

thomasfredericks, 2025-06-14:
`0xFFFFFFFF = 4294967295`

But `fixedTofloat(0.5)` is declared as `0.5 * 0xFFFFFFFF` = `2147483647.5`

The result is NOT `0x80000000` (2147483648)  but one bit shy. Is this an issue?

thomasfredericks, 2025-06-14:
Would Q1.31u be better? 

Float have ~6–7 decimal digits
Q0.32u have ~9.6 decimal digits
Q1.32u have ~9.3 decimal  digits

Please note that most of this information is AI generated, but it seems to compute.

| Operation             | Float (Software FPU) | Q0.32 (Integer math) | Speedup     |
|-----------------------|----------------------|-----------------------|-------------|
| Addition / Subtraction| ~20–40 cycles        | 1–2 cycles            | 10×–40×     |
| Multiplication        | ~100–200 cycles      | 2–5 cycles (via `uint64_t`) | 20×–100×    |
| Division              | ~150–300 cycles      | 10–30 cycles (if needed)   | 5×–20×      |
| Memory use            | 32 bits              | 32 bits               | —           |

thomasfredericks, 2025-06-14:
Here is some Q1.32u code for refenrece:
```cpp
int32_t q1_31_mul(int32_t a, int32_t b) {
    return (int32_t)(((int64_t)a * (int64_t)b) >> 31);
}

int32_t half = 0x40000000; // 0.5 in Q1.31
int32_t result = q1_31_mul(half, half); // Should be 0.25 = 0x20000000

int32_t one = 0x7FFFFFFF; // Max positive value ≈ 1.0 in Q1.31
int32_t half = 0x40000000; // 0.5 in Q1.31
int32_t result = q1_31_div(half, one); // Should be ≈ 0.5 = 0x40000000
```

sofian, 2025-12-11:
I will circle back to the original question:

... "The result is NOT 0x80000000 (2147483648) but one bit shy. **Is this an issue?**"

I am not 100% clear on whether it is an issue or not. What is the problem exactly?

thomasfredericks, 2025-12-16:
The issues seem to be fixed with the new code

closed 2025-12-27 by sofian