SofaPirate/Plaquette #145: Improve documentation by providing more images

sofian, 2025-06-20:
The documentation needs more images. Circuit diagrams would be really cool to have so that people can easily reproduce the examples.

Also some evocative visuals could be interesting to have (eg. actual images of LEDs, etc.) Look at online or PDF guides or manuals such as https://docs.arduino.cc/learn/starting-guide/getting-started-arduino/ Sometimes you can even directly borrow images from these sites (verify if they are public or creative commons).

www-mariie, 2025-07-29:
@sofian I have two questions regarding additionnal images.

1. How should i name these files / structure them in the image folder? Is there a nomenclature i should follow? Here is an exmaple of what I did with the circuit images, directly placed in the "Plaquette/docs/images" directory:

[image]

2. Should I compress the image files to smaller sizes for better web performance? If so, what size am I aiming for?

sofian, 2025-08-17:
1. They need to be placed in the "Plaquette/docs/images" directory. There is no specific nomenclature, I think what you are proposing is perfect.
2. Yes you should definitely aim for reasonable size. I think no more than 1920x1080.

closed 2025-11-18 by sofian